angelsix/solidworks-api #6: Tutorials\02-WpfAddIn Solution is not compileable

guxl, 2017-09-19:
the error message show that the "NInject" package can not resolved.

when I reinstall the "NInject" package and recomplie.

the error message show that 
```
错误	CS1705	标识为“AngelSix.SolidDna, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null”的程序集“AngelSix.SolidDna”所使用的“SolidWorks.Interop.swpublished, Version=25.1.0.49, Culture=neutral, PublicKeyToken=89a97bdc5284e6d8”版本高于所引用的标识为“SolidWorks.Interop.swpublished, Version=24.3.0.57, Culture=neutral, PublicKeyToken=89a97bdc5284e6d8”的程序集“SolidWorks.Interop.swpublished”	SolidDna.WpfAddIn	C:\Users\crayfish\Downloads\solidworks-api-master\solidworks-api-master\Tutorials\02-WpfAddIn\SolidDna.WpfAddIn\Class1.cs	10	活动的
```
That is, the reference to `SolidWorks.Interop.***` of `SolidDna.WpfAddIn` project is different from `AngelSix.SolidDna` project. 
When I resolved this question and recomplie， the error message show that
```
错误	CS0534	“MyAddinIntegration”不实现继承的抽象成员“AddInIntegration.PreLoadPlugIns()”	SolidDna.WpfAddIn	C:\Users\crayfish\Downloads\solidworks-api-master\solidworks-api-master\Tutorials\02-WpfAddIn\SolidDna.WpfAddIn\Class1.cs	10	活动的
```
and that
```
错误	CS0246	未能找到类型或命名空间名“ISolidPlugIn”(是否缺少 using 指令或程序集引用?)	SolidDna.WpfAddIn	C:\Users\crayfish\Downloads\solidworks-api-master\solidworks-api-master\Tutorials\02-WpfAddIn\SolidDna.WpfAddIn\Class1.cs	25	活动的
```

angelsix, 2017-09-19:
1. Make sure you re-install NInject which it looks like you have
2. Check your References for each project, make sure all the SolidWorks interop Dlls are set to Embed = False in the properties
3. Make sure both projects reference the same SolidWorks Dlls
4. Try compiling the source code example directly here https://github.com/angelsix/solidworks-api/tree/develop/Tutorials/06-Exporting. Does that work?

guxl, 2017-09-19:
1. NInject project is same to origin one.
2.  Embed = False.
3. Both projects reference the same SolidWorks Dlls
4. https://github.com/angelsix/solidworks-api/tree/develop/Tutorials/06-Exporting. This project works well.

But the question is still exists.

guxl, 2017-09-19:
I can not find `ISolidPlugIn` in whole solution. I am not sure this interface is provided  either by solidworks interop DLLs or by `AngelSix.SolidDna` project .

angelsix, 2017-09-19:
The old tutorial 02 WpfAddin is different from the newest code. If you are using some code from the new project and some from the old tutorial it will not work. 

I've updated the older tutorials to the new style now so just get it here:

https://github.com/angelsix/solidworks-api/tree/develop/Tutorials/02-WpfAddIn

closed 2017-09-19 by angelsix

guxl, 2017-09-19:
All tutorials work well now!

Mussawer, 2020-12-07:
Hello, 
I am not able to register my dll because of sldwork.interop.swpublished in the references of solution but when i remove its reference I am able to register it using the tool also through command line. 

By doing this my addin is not loaded in soliworks neither I am able to debug it through visual studio