angelsix/solidworks-api #11: How to change toolbar and menu visibility?

brinkdinges, 2017-12-14:
Hi Luke,

Awesome work on keeping this project going and expanding. I just created my first menu using your CommandManager. How do I change the HasToolbar and HasMenu setting after I called the CreateCommands method? I can change the parameter on the return value (type CommandManagerGroup), but that doesn't change the result. I guess I have to update SolidWorks as well or reactivate?

As an alternative, maybe we can add those two parameters to the CreateCommands and CreateCommandGroup method and to the CommandManagerGroup constructor?

angelsix, 2017-12-14:
I haven't played to see if simply recalling Activate works.

If not it would need removing and recreating. 

If it is the latter you could do a remove/reactivate in the property setter.

brinkdinges, 2017-12-15:
I've tried adding a Reactivate method that just calls Activate on the base object of the CommandManagerGroup, that doesn't change the toolbars. 

I don't think we need to be able to change these settings on the fly after initialization. I think few users would change these settings while SW is running, the majority would initialize it with their preferences of HasToolbar and HasMenu and never change this. For that reason, I'd like to propose adding it to the CommandManagerGroup constructor. 

For now I've added this for myself. Just let me know if I should create a pull request.

angelsix, 2017-12-17:
Yes add it to constructor and I'll merge

closed 2017-12-30 by angelsix