angelsix/solidworks-api #12: How and when to use UnsafeObject?

brinkdinges, 2017-12-14:
Here's another subject that I don't fully understand yet. I was hoping you could shed some light on it. 

My project interacts with SolidWorks a lot and most of it uses original SW API because the functionality isn't included in SolidDna yet. Right now I have a SldWorks object with a getter that returns Dna.Application.UnsafeObject. I then use ActiveDoc to get the current model.

1. Is that the proper way to access SolidWorks directly?
2. It usually gives an error when I close SolidWorks, so I guess I have to dispose of the object? How do I do that and when?

Thanks again,

Peter

angelsix, 2017-12-14:
1. Correct 
2. Yes sounds like you just need to dispose somewhere. Look how we do it with the SolidDnaObject just Release com basically

closed 2017-12-30 by angelsix

brinkdinges, 2018-01-21:
I'm still having trouble understanding what I need to do here. This is how I can recreate the exception:

1. Open a part
2. Create a drawing using my add-in
3. Close SolidWorks while both documents are open

The exception that I get is: "COM object that has been separated from its underlying RCW cannot be used. The COM object was released while it was still in use on another thread." on line 335 of SolidWorksApplication.cs. It's the line that checks for any open files after 200ms:

if (mBaseObject?.ActiveDoc == null || mBaseObject?.GetDocumentCount() == 0)

The Dispose method on the SolidWorks application is called after the 200ms starts and before the 200ms is up. Has SolidWorks already disposed of objects that we still need?

angelsix, 2018-01-21:
Try now with this fix https://github.com/angelsix/solidworks-api/commit/da7a3f6b7339562cf9732878505112b0a9f5f00d

brinkdinges, 2018-01-21:
That makes sense, thanks. 

I ran into a few weird/missing events as well. There is an event that fires when a drawing sheet is renamed, but it only fires in 1 of 3 ways it can be renamed. 

I decided to stop using the sheet name as an identifier. SolidWorks uses names as identifiers all the time, which even causes a random model to be closed when there are two with the same name... Jeez.

brinkdinges, 2018-01-21:
It didn't work yet because the disposing starts when the timer is running. I tried moving it to after the delay and that did work :)

angelsix, 2018-01-21:
Oops yeah I did that in wrong place. Just on the go now. Can you move it and commit a pull

brinkdinges, 2018-01-21:
I spoke too soon. Same error occurred once again. I'll fiddle with it some more and create the pull request when it works reliably.

brinkdinges, 2018-01-21:
This is strange. It still tries to reload the model info even though Disposing is true. I even got it to go into ReloadActiveModelInformation() and then throw, not in the line before where it usually throws. Those two lines must be a millisecond or less apart.

angelsix, 2018-01-21:
Wrap the whole Dispose function of our SolidWorks object in a lock and then wrap our code after the delay 200 in that same lock then it should be ok