angelsix/solidworks-api #15: Event when closing last document

brinkdinges, 2018-01-06:
Hi Luke,

I've been trying to find a way to receive an event when the last document is closed. As you have written in SolidWorks/Application/SolidWorksApplication, there is no special event for when the last document is closed. 

`private void ActiveModel_Closing()
        {
            if (mBaseObject.GetDocumentCount() == 1)
                Task.Run(async () =>
                {
                    // Wait for it to close
                    await Task.Delay(200);

                    // Now if we have none open, reload information
                    if (mBaseObject?.GetDocumentCount() == 0)
                        ReloadActiveModelInformation();
                });
        }`

The situation where GetDocumentCount is 1 is very rare. It only happens when a single part or a lightweight assembly is open. If a part has an external reference (for example a forming tool) the count is >1. With assemblies, each component counts and a drawing opens it reference part/assembly as well. 

I thought we could check if ActiveDoc is null after each file close and after a little delay. I've checked the code below and it works with an assembly with 500 unique components as well. Closing took a few seconds, but the Reload method is still reached.

Apparently ActiveDoc can be null in rare other occasions, but most likely not just after closing the previous file.

`private void ActiveModel_Closing()
        {
                Task.Run(async () =>
                {
                    // Wait for it to close
                    await Task.Delay(200);

                    if (mBaseObject.ActiveDoc == null)
                        ReloadActiveModelInformation();
                });
        }`

What do you think? Shall I make a pull request?

angelsix, 2018-01-06:
The reload will always happen for new files getting open or changed.. that is still a correct assumption?

If so we are only trying to detect no documents left open still? If that is true, what about this instead?

````
private void ActiveModel_Closing()
{
    Task.Run(async () =>
    {
        // Wait for it to close
        await Task.Delay(200);
        if (mBaseObject?.ActiveDoc == null || mBaseObject?.GetDocumentCount() == 0)
             ReloadActiveModelInformation();
         });
     }
}
````

That way we cover our bases for both of the checks?

brinkdinges, 2018-01-06:
ReloadActiveModelInformation is still called when a file is opened or the user switches between files. It's still called once. I was only able to trigger it twice by closing a non-active document via the Window toolbar. Solidworks apparently activates and closes that document.

I did a few tests when closing the last file. After 200ms:
- Single simple part: ActiveDoc null, count 0
- Single drawing: ActiveDoc null, count 0
- Simple assembly: ActiveDoc null, count 0
- Big assembly: ActiveDoc null, document count hasn't changed, still 500

So it seems to work reliably when we implement both checks. I'll create the pull request.

closed 2018-01-20 by angelsix