angelsix/solidworks-api #16: Assembly not found when not copied to Solidworks folder

brinkdinges, 2018-01-09:
Hi Luke,

I've been trying to add Log4net to my addin so I can write to a log file and write to a SAAS platform. I'm trying Loggly right now, they have a Nuget package for log4net. 

The [default ](https://github.com/loggly/log4net-loggly) way of setting this up would be to add a log4net section to app.config. That didn't work. Is that because we run on the SolidWorks thread and not as a separate application? I read that app.config isn't completely read when you don't run your own application and I also wasn't able to read sections of the config file when running.

I got writing to a file working by configuring log4net in code using [this](https://blog.frogslayer.com/configuring-log4net-in-code/) sample. But when I add a reference to the log4net.loggly appender, I get a FileNotFoundException and it can't find log4net.dll. It only works when I copy that file to the SolidWorks folder. Is this expected behavior or is there a reference not working as expected? The log4net-loggly code is on Github as well (see first link) so I can change anything I need to. I just have no idea what I should be looking for.

dhwed, 2018-01-09:
If you're running in the SolidWorks AppDomain, then you'll probably have to tell that AppDomain to also search for assemblies in your plugin's directory.

Take a look at [this stackoverflow answer](https://stackoverflow.com/a/1373295/2798869) for details on how to do this.

angelsix, 2018-01-09:
I changed it back to run in the SolidWorks AppDomain (so like a regular thing) a while ago https://github.com/angelsix/solidworks-api/commit/fecfa3ec2905713c649b443811fb851892263793

Without seeing the exact code we could go back and forward. If you sent a simple example project I could take a look at the underlying issue, however it should not be any different than a standard project in terms of setup

brinkdinges, 2018-01-09:
Thank you both. Now I know in which direction I should look. Switching to the SW AppDomain already saved me a major headache because I can now properly work with drawings again. 

I'll create a test project for you to look at :)

brinkdinges, 2018-01-09:
Here you go:

https://github.com/brinkdinges/appDomainSwitchTest

You might have to replace the Solidworks references for the versions that you're currently using. 

When this addin loads it writes to the log file specified in Logging.cs. It fails when you uncomment the Loggly appender. I removed my personal input key to keep it private, you can use this temporary one if necessary: e2baf18a-fcac-4f21-ad27-ec83d17fb2fa

dhwed, 2018-01-09:
Try adding this into your `Integration.PreConnectToSolidWorks()` function:

```csharp
if (!PlugInIntegration.UseDetachedAppDomain) {
    string dir = Path.GetDirectoryName(Assembly.GetExecutingAssembly().Location);

    AppDomain.CurrentDomain.AssemblyResolve += (sender, args) => {
        string assemblyPath = Path.Combine(dir, new AssemblyName(args.Name).Name + ".dll");
        return File.Exists(assemblyPath) ? Assembly.LoadFrom(assemblyPath) : null;
    };
}
```

brinkdinges, 2018-01-09:
That seems to have worked, thanks so much David! So this resolves all assemblies that fail to resolve on the first try?

dhwed, 2018-01-09:
My understanding is that if it can't find the assembly in the SolidWorks folder under Program Files or in the GAC, then it will fire that AssemblyResolve event.

If you're going with this approach, be aware of the potential issues discussed [in this article](https://blogs.msdn.microsoft.com/abhinaba/2013/09/29/custom-resolution-of-assembly-references-in-net/).

brinkdinges, 2018-01-09:
Point 3 seems to be the most relevant, right? So don't try to load everything, just known assemblies. I'll keep it in mind and read the article a few times more.

dhwed, 2018-01-09:
Yeah, it seems like it is safer to only resolve the specific assemblies you actually need.

angelsix, 2018-01-10:
So SolidWorks loads your add-in, hence the assemblies present in your add-ins folder wont get automatically loaded as you would expect if you had another exe style program.

What this code is doing now is explicitly searching and effectively adding the add-ins dll folder as a resolve folder making it act similar.

To be honest I think this style of resolve should be baked into the SolidDna by default to provide the similar experience to a normal program... what do you guys think?

brinkdinges, 2018-01-10:
I think it would be a great feature for SolidDna as well. You invite even more people in when they don't have to worry about these things. It really lowers the step of going from macros to addins.

dhwed, 2018-01-10:
You have a good point that just loading everything isn't any different than the normal behavior when you're running multiple plugins all in the SolidWorks AppDomain.

So, having SolidDna add this event handler for us makes sense, but only when `PlugInIntegration.UseDetachedAppDomain` is `false`.

angelsix, 2018-01-12:
Yep I'll get that added when back from work trip next week unless you guys want to do a pull request in the meantime

angelsix, 2018-04-26:
This is done. If running in PlugInIntegration.UseDetachedAppDomain = true, you can now set PlugInIntegration.ResolveAllAssemblies to true and it will resolve all assembly

https://github.com/angelsix/solidworks-api/commit/94c3c17a596af37a956cb7bcdaa6a9a06749c398

closed 2018-04-26 by angelsix