angelsix/solidworks-api #20: Error cleaning up after first save

brinkdinges, 2018-01-12:
I'm in the process of adding Save and Save As buttons to my application. When I save a model using for the first time (using ModelDocExtension.SaveAs), Model.cs reloads the model data and tries to destroy all previous references. This results in the exception "An attempt has been made to free an RCW that is in use" by SolidDnaObject.Dispose: Marshall.FinalReleaseComObject when mBaseObject is a modelExtensionClass.

Is this expected behavior? How can I prevent this from happening? I tried adding the save functionality to the SolidDna ModelExtension, but that gives the same error, I assume because it uses the same object.

For now I have disabled the ReloadModelData on first save.

angelsix, 2018-01-20:
See if the latest code works now... seems to for me https://github.com/angelsix/solidworks-api/commit/48e72d83c97ec99d3c3888b30e3e0191e6df76c4

You can now just do ActiveModel.SaveAs as I did all the code in there for you too

brinkdinges, 2018-01-20:
Works for me as well, thanks.

closed 2018-01-20 by brinkdinges