angelsix/solidworks-api #24: Addin fails to install after obfuscation

brinkdinges, 2018-03-01:
Hi Luke,

I was wondering if you could point me in the right direction here. I'm creating an installer of my addin using NSIS. The installer calls regasm and registers the app so it can be removed using Control Panel. That all works. 

Now I'd like to obfuscate my dlls, I'm trying Obfuscar for that. The new dlls get built and I've checked with .Net Reflector to make sure everything is obfuscated. It seems that all private methods and classes are obfuscated and all public method and class names remain the same. Great. 

It fails when I replace the non-obfuscated dll with the obfuscated ones and try to install again. Sometimes the add-in shows up as only my company name, which I guess is a ToString() of the dll name or something. Sometimes the add-in doesn't even show up. 

Any idea what could go wrong here? I think I should be able replace the original dll with the obfuscated one without problems? It doesn't have its location built in, right?

angelsix, 2018-03-01:
regasm will write registry entries to your registry based on the assembly.

It should write an entry in `HKLM\Software\Classes\{yourProgId}`

If you don't set one this is random. You can set one using an attribute on the class that has the `[ComRegisterFunction()]` inside it like this:

```
[ProgId("yourProgIdGuid")]
public class MyClass
{
    [ComRegisterFunction()]
    private static void ComRegister(Type t) { ... }
}
```

It will also set inside that folder the class ID which you can also set on the same class via:

```
[Guid("yourClassIdGuid", ComVisible(true)]
```

It then also adds the class ID entries into your registry in `HKLM\Software\Classes\CLSID\{yourClassId}` and inside there a bunch of stuff.

If you look there you will get an idea of what is going wrong.

I don't use regasm for my installer I use Wix and manually add every registry entry that regasm would create.

brinkdinges, 2018-03-01:
Awesome, thanks for the detailed response. I will dive in to this and see if I can figure it out.

SolidDna's AddinIntegration.cs has methods with the ComregisterFunction and ComUnregisterFunction attribute. I already have a class that inherits from UserControl and  ITaskPaneControl with the ProgId attribute like "Company.ProductName", as shown in your second SolidDna video. 

I don't have the Guid attribute yet, I see it's also in the example that I received from Solidworks. So that has to be added to the class that inherits from the AddInIntegration? Even though the assembly already has a Guid mentioned in the assembly settings, I assume because SW addins are different from a standalone program? Does that suffice for the code part?

Would you advise against regasm in the installer? I should be able to add the entries manually with NSIS as well. I heard Wix has a really steep learning curve.

angelsix, 2018-03-04:
The Guid is for the class. The guid in the project properties is for the assembly. Two different things. So the Guid attribute is used for the COM registration, nothing specific to SolidWorks, just specific to COM.

I would not use regasm on an installer, just manually add all the registry entries yourself that you know would be generated (just monitor for registry changes using software and then clone those entries with the installer, thats what I do with my SW installers).

closed 2018-03-04 by angelsix