angelsix/solidworks-api #25: Plugin loads twice when two addins use SolidDna

brinkdinges, 2018-03-14:
Hi Luke,

I've created an addin for a customer and we ran into an issue when two of our addins are loaded at the same time. SolidDna is called for each addin, but the first addin to load is loaded twice.

I think this happens because PluginIntegration.PlugInFolder is set to typeof(PlugInIntegration).CodeBaseNormalized(); and only a single SolidDna DLL is loaded. It then finds the same code base twice and continues loading the first addin twice.

How can we prevent this? Can we pass the location of the plugin in a constructor somewhere?

I'll create a second SolidDna DLL with a different name for now.

angelsix, 2018-03-15:
Can you create me and zip me up an example of 2 add-ins so I can just install them and debug. Should be able to fix it in few hours tops

brinkdinges, 2018-03-15:
Sounds good. I'll make two and send em to you.

angelsix, 2018-04-26:
Did we resolve this?

brinkdinges, 2018-04-26:
No not yet, just didn't get around making two projects for you. I'll put it in my calendar for this weekend, because it would be great if we can support multiple addins running simultaneously.

brinkdinges, 2018-04-29:
I just sent you the two samples :)

angelsix, 2018-05-02:
So issue one is your samples both had the same Guid for the AddinIntegration. Change that to another like this::

```
    [Guid("D99D8F78-7D80-42BB-B16E-8D2015E83887"), ComVisible(true)]
    public class AddinTwo : AddInIntegration
```

Then I did some digging and there is a very strange situation I think SolidWorks (and not SolidDna) is responsible for. I'm not 100% sure right now and will dig more, but I commented my findings here https://github.com/angelsix/solidworks-api/commit/de86a19f3c8144f2e8563b19dc0b90b73c159bbe#diff-ac3158a86f81f0098a2f2bd572dc9a9cR395 

Turns out that if SolidWorks loads a file/dll with the same name, it seems to just re-use that dll for any future attempts that reference a dll of the same name, even if it is a different assembly version, file version, GUID or date stamp. So once add-in one loaded, add-in two would make use of the dll in add-in ones folder, not in its own, and it would also be the same instance of that dll too, meaning all static and in-memory values inside that dll (like the `PlugIns` list) were shared, and the `PlugInFolder` referenced the location of the wrong dll.

I've added a specific fix to that now to solve that one issue (https://github.com/angelsix/solidworks-api/commit/de86a19f3c8144f2e8563b19dc0b90b73c159bbe) but I have not fixed or found a direct cause or solution to the underlying issue (hence the warning added in code to be aware of for now).

It doesn't seem to cause any other issues right now with how we use SolidDna so its an ok solution, but I will try and find the real cause. Feel free to investigate yourself if you would like. The easy way to test what dll is loaded is inside say `PluginIntegration` class in `ConnectedToSW` you can do `var myPath = typeof(PlugInIntegration).Assembly.Location` and it will tell you the full path of the dll. Right now its always using the first loaded dll, even with the fix, but we bypass that issue with the solution I did in the fix.

angelsix, 2018-05-02:
Oh and the fix to using the PluginFolder to find your current folder I fixed with the extension helper I made here https://github.com/angelsix/solidworks-api/commit/de86a19f3c8144f2e8563b19dc0b90b73c159bbe#diff-7be263e76e0501a832399212932e34d6R16 so you can call `this.AssemblyPath()` to get the path to the assembly where the class you are in is located, and you use it like this

```
public override void ConnectedToSolidWorks()
{
    _taskPane = new TaskpaneIntegration<TaskPaneHost2>
    {
        Icon = Path.Combine(this.AssemblyPath(), "ImageTwo.png"),
        WpfControl = new TaskPaneWindow2()
    };

    _taskPane.AddToTaskpane();
}
```

closed 2018-05-02 by angelsix

brinkdinges, 2018-05-03:
Thanks for digging into the issue, I will give the fix a go. I missed the identical GUIDs in the example, but the strange behavior you found is identical to what I've seen. 

Somehow it doesn't surprise me that SolidWorks can't handle two files with the same file name. This stupid behavior has caused me problems before with drawings and parts/assemblies. I was able to create two models with the same title, after which calling the close command would close a random one. A unique model ID would be so great.