angelsix/solidworks-api #28: Macrofeature suport

emersonbottero, 2018-04-26:
Add suport or example on How to workshop with macrofeatures.

angelsix, 2018-04-26:
Can you elaborate a bit more on what you are after?

emersonbottero, 2018-04-26:
Something like that.
https://www.youtube.com/watch?v=I-cIGMFSBOg

You have to create a ISCOMServre project or class and implement edit, rebuild and security functions.

macrofeatureClass needs to have registry keys in HKCR\CLSID\{1FEC45A4-8300-4FD2-AFC4-ED4952A6F4F5}\InProcServer32\CodeBase

Value should be file:///PathToYourDLL

(Default) should be MSCOREE.DLL

Overview of MacroFeatures
http://help.solidworks.com/2014/english/api/sldworksapiprogguide/Macro_Features/Overview_of_Macro_Features.htm?id=277bd61b627d485493b0c6c08c8bee27#Pg0

Most of the time you have to add a PropertyManagerPage or interface to set or get data from macrofeature data object. I Have done that With PropertyManagerPage where I get all public fields in it and add to the MacrofeatureData  ,if there are other fields needed for the interface but not for the MacrofeatureData turn them in to property. To be Able to use it in more than one PMP I create an MacroFeaturePMP class that all other must inherit and it has only 2 properties, BaseName and IconFiles.  That should occur in the Edit callback.

Them you use a Solidworks Modeler to add or remove solids from your Model based in the data set in the Interface. That should occur in the Rebuild callback

You can check the parameter set in the interface before rebuild to make sure it is all correct, like someone put 0 in radius to model a cylinder, and warn them. That Should occur in Security.

I did it one that is not working , the model is created by the modeler but it never shows in the screen and I think is because I put the COMServer together with the swAddin class. 

MacroFeatures are super Cool and seems part of Solidworks when they are finished.

angelsix, 2018-04-26:
Thats actually something I didn't even know about! That's really cool. I will make that a feature to do a video on in the near future for sure. I'll keep this issue open until then

emersonbottero, 2018-04-26:
I can send you the code I already did if you want.

angelsix, 2018-05-05:
Thanks I'll look into this soon as I get time