angelsix/solidworks-api #29: wrong Text Format on

emersonbottero, 2018-04-27:
[image]

should be versão 4.7

angelsix, 2018-04-27:
This is the error message of regasm.exe when the dll cannot be successfully registered. Nothing to do with this source code

closed 2018-04-27 by angelsix

emersonbottero, 2018-07-12:
Are you sure?In the terminal it works well. You change the text and I don't need it now, maybe when there is an error.
In other situations we set
 Encoding.GetEncoding("iso-8859-1")
or
Encoding.GetEncoding(28591)
is it possible?

[image]

angelsix, 2018-07-12:
The error we just fixed for detecting success when non-english version was called will of likely fixed this same issue