angelsix/solidworks-api #30: Can't load the examples

emersonbottero, 2018-04-27:
The example 1 loaded ok.
the example 2 brings the error
Attempt to load an assembly from a network location that would cause the assembly to be in safe mode in earlier versions of the .NET Framework. This version of the .NET Framework does not enable CAS policy by default, so this load can be dangerous. If it does not intend to place the assembly in a restricted area, enable the loadFromRemoteSources option. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.
I added a configuration folder with

<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <runtime>
    <loadFromRemoteSources enabled="true"/>
    <legacyCasPolicy enabled="false" />
  </runtime>
</configuration>

But I can't test it with Solidworks 2017.

angelsix, 2018-04-27:
Don't put the source code on a network drive. You cannot have dll's registered for COM when they are on a network drive

closed 2018-04-27 by angelsix

emersonbottero, 2018-04-27:
I'm not in a network!!

angelsix, 2018-04-27:
[image]

Works fine. Must be your machine/network setup. Any more info?

emersonbottero, 2018-04-27:
You are Right. You need to do it in every single dll.
[image]

lowEagle, 2018-05-21:
Sadly I have a very similar problem. The very first tutorial works well (written "by hand" simultanious to watching the youtube-video and also via import of the current directory copied from this repository). 

However, as soon as I try to use your Framework (Tutorials 02-07) the addin will be registered (-> will be shown in the Addin-Menu) but not loaded. I tried to debug using Visual Studio and the very same break-point as shown in the 7th tutorial but the stop will never be reached. 

I'm on the current version of Win10, x64. The files are stored locally and all libraries are manually set like  @brinkdinges and @emersonbottero described in their previous posts. Unfortunately the project I'm working on forces me to use the Solidworks 2015-16 Edition. Anyways today I removed that and tried the current Student Edition (2017-18) instead - without any changes. 
I was hoping the use of the NuGet-Package will solve this issue but that won't work either.

Tomorrow I'm gonna try to register the compiled files on another computer. Any other suggestions?

emersonbottero, 2018-05-21:
Have you change all referenced dll to the 2015-16?I think if you just overwrite the ones in the example you shouldn't had to change the code.

lowEagle, 2018-05-21:
That was one of the first approaches, but currently it will lead to a mismatch between them and the ones includes by the NutGet-Packet (Translated Build-Error):
Error CS1705 Assembly "AngelSix.SolidDna" with identity "AngelSix.SolidDna, Version=1.0.0.6, Culture=neutral, PublicKeyToken=null" uses "SolidWorks.Interop.swpublished, Version=26.0.1.1, Culture=neutral, PublicKeyToken=89a97bdc5284e6d8" with a higher version than the referenced assembly "SolidWorks.Interop.swpublished" with identity "SolidWorks.Interop.swpublished, Version=25.1.0.49, Culture=neutral, PublicKeyToken=89a97bdc5284e6d8".	SolidDnaNuGet	C:\Users\fabian\source\repos\07-SolidDnaNuGet\SolidDnaNuGet\SolidDnaIntegration.cs	

All libraries are set to be not included (Embed interoptypes False). 

I will do another test with the previous, non-Nutget, example.

angelsix, 2018-05-21:
There is no need to use older DLL files use the 2018 files they work for older versions. 

Further more just open Visual Studio go to Tools Extensions and search for SolidDNA then install that. The go to Project New and you can see a SolidDNA project all setup for you https://youtu.be/Jhi1WFp47Qk

angelsix, 2018-05-21:
For reference here is how to use NuGet package reference https://youtu.be/r8e3xFlmZ_o

lowEagle, 2018-05-22:
First of all thank you @angelsix for the work you are doing.

I was able to get the past tutorials running using the 2018 DLL files and resetting some of the settings. However, the project based on the NuGet package still won't load.

I think I've set everything like desribed in your videos. The build works finde (except two warning regarding a versioning conflict of the files Microsoft.Extensions.Configuration.Abstractions and Microsoft.Extensions.Logging.Abstractions). The registration via your tools returns no error. I even tried registering manually via cmd-line. On SW startup the addin appears in the loading screen. Strangly there is no entry in the SW log: No error, no "LOAD_ADDIN". Like before it is not shown as activated in the addins panel and cannot be activated manuelly there.

For reference I took some screenshots that you can find attached. Maybe I'm overseeing something.
The RestoreProjectStyle is set to PackageReference (which can now also be done globally in the NuGet section of the Visual Studio options. However, it has also been done in the .proj-file).
The projectname shown in the screenshots is from my last attempt, using the exact same name like in your videos won't change anything. Message shown by your registration-program is the same as when registering all other projects; the last line says that the types have been registered successfully.
[SDNA_NuGet-Project_Screenshots.zip](https://github.com/angelsix/solidworks-api/files/2027733/SDNA_NuGet-Project_Screenshots.zip)

brinkdinges, 2018-05-22:
The last time I checked I also couldn't get the addin running when using the NuGet package. The latest clone of the repo (9ff89d1772f2c294e0b188ff9f27628286c0e319) worked, which is why I didn't reopen my issue https://github.com/angelsix/solidworks-api/issues/31 immediately. But the NuGet package from the same date doesn't work.