angelsix/solidworks-api #32: AngelSix.SolidDna nuget package

doronkind, 2018-05-28:
Luke, good morning!

I want to ask you about your experience in SolidWorks add-in writing. I've found your YouTube lessons and according to them tried to implement several add-ins. Later I've found the Nuget package AngelSix.SolidDna and tried to add this, cause it incapsulate great sw API features. 

Unfortunately I run into trouble, cause

1. When I add Nuget package instead of the reference to the SolidDna project I can't download add-in, but it can be solved by adding SolidWorks.Interop.* dlls (But do we need to add them directly, if Nuget packege incapsulate features)

2. When I tried to debug add-in 've notice, that method "ConnectedToSolidWorks" has never been called. But whan I return the reference instead of the Nuget package it works.

Could you advice me the best approach according to your API and it would be great if you tell me about the Nuget package state, does it stable and can be used?

P.S. I am working with the SW 2016

angelsix, 2018-05-29:
The main difference will be you need to edit your csproj file and add this https://github.com/angelsix/solidworks-api/blob/develop/Tutorials/07-SolidDnaNuGet/SolidDnaNuGet/SolidDnaNuGet.csproj#L14  `<RestoreProjectStyle>PackageReference</RestoreProjectStyle>`

You can see it here https://www.youtube.com/watch?v=r8e3xFlmZ_o

Also now there is a simple template you can download and use so you can do New Project - SolidDNA to create a project for you. https://www.youtube.com/watch?v=Jhi1WFp47Qk (near last 5 minutes of video)

Let me know if you still have issues

doronkind, 2018-05-29:
@angelsix Unfortunately it didn't help! 

I've tried to use your tutorial example with nuget in this way: 
1. Add references and set Embed interop types to false;
2. Restore Nuget package.
3. Build project and load add-in
4. Start debuging in SW
5. Add breakpoint in the "ConnectedToSolidWorks" method
6. Enable add-in

but the method ConnectedToSolidWorks still has never been called. BTW method PreConnectToSolidWorks called. 

Is the add-ins menu a see the following image: 

[image]

Load time always "--" and it seems, that the add-in can't connect

angelsix, 2018-05-29:
If you have created it from the project new, you don't have to add references, they already exist. So are you sure you have tested it from the new project?

doronkind, 2018-05-29:
I did the same as in this video: https://www.youtube.com/watch?v=r8e3xFlmZ_o

angelsix, 2018-05-29:
Ok to confirm if its your project or something else, just go to `Tools - Extensions` search for `SolidDNA` install that, then go `Project - New - SolidDNA - Blank SolidDNA project`, build and register that, run and see if that works

doronkind, 2018-05-29:
Unfortunately, the same result, even if I use SolidDNA template. Maybe it is because of the 2016 version of the SW?

angelsix, 2018-05-29:
Possibly a SW 2016 issue then. I'm away on holiday at moment but will install SW2016 when I can over next day or so and test. 

In the meantime can you just reference the SolidDNA source code instead and step into it line by line or turn on Ctrl + Alt + E and check the second box down then debug it and see if anything crashes. The template you created details how to debug SolidWorks and also the boot order of a SolidDNA template so you could see at what point it fails. 

If not don't worry I'll debug it myself shortly but I may not see any issue even if so. Hopefully I will see an issue on SW2016 and will fix. 

Also can you download a trial of SW2017 or 2018 to test it works on that?

doronkind, 2018-05-29:
Sure, I am going to try with SW 2018, hope it will be succesfull! Thank you for help!

doronkind, 2018-05-29:
@angelsix I've checked the benaviour with 2018 version and it still wrong, so, maybe this problem related with my computer. I am going to try the same on another one and after this tell you about results.

doronkind, 2018-05-29:
@angelsix BTW could you describe how icons list path works for commands. Is it just several .png joined into one?

angelsix, 2018-05-29:
It's it's in there and hitting preload it's not your PC. Do you have multiple SolidDNA addins installed?

doronkind, 2018-05-29:
Maybe yes, but I'am not sure. I've tried to clean this branch in the registry:

[image]

angelsix, 2018-05-29:
For the icons yeah look in source it describes it https://github.com/angelsix/solidworks-api/blob/develop/SolidDna/AngelSix.SolidDna/SolidWorks/CommandManager/Group/CommandManagerGroup.cs

angelsix, 2018-05-29:
Can you debug the startup flow. The flow is explained here. See at what point it stops loading https://github.com/angelsix/solidworks-api/blob/develop/Templates/SolidDna.Blank/SolidDNA.Blank/SolidDnaIntegration.cs

angelsix, 2018-05-30:
Ok fresh PC, fresh install, from template. Found the issue... investigating now
[image]

doronkind, 2018-05-30:
Sounds good, If you need more questions ready to help

angelsix, 2018-05-30:
Ok so should be fixed now. Underlying issue was simply an old version of Dna Framework being referenced that isn't available. I've updated SolidDna https://github.com/angelsix/solidworks-api/commit/0d2b45255cb2a226071c73ef5abb6e5aa183fe70

Then a new NuGet package is now available for SolidDna 1.0.0.9 https://www.nuget.org/packages/AngelSix.SolidDna/

Finally, I've updated the template https://github.com/angelsix/solidworks-api/tree/develop/Templates/SolidDna.Blank/SolidDNA.Blank it now has an appsettings.json file that sets the logging to trace to output everything, and then you now get full console, debug (and file https://github.com/angelsix/solidworks-api/blob/develop/Templates/SolidDna.Blank/SolidDNA.Blank/SolidDnaIntegration.cs#L99) log of the startup process so you can see it more clearly and debug any issues in future.

[image]

If you still get issues now after using the new SolidDNA template 1.0.1 https://marketplace.visualstudio.com/items?itemName=AngelSix.soliddna-blank or from the NuGet package, let me know. 

Also let me know if it now works.

angelsix, 2018-05-30:
And if you add the file logging https://github.com/angelsix/solidworks-api/blob/develop/Templates/SolidDna.Blank/SolidDNA.Blank/SolidDnaIntegration.cs#L99 you will get a file log output like this too

[image]

doronkind, 2018-05-31:
@angelsix I've added SolidDna 1.0.0.9 and right now it works as expected. Thanks for the help, your SW API is really great!

closed 2018-05-31 by doronkind

angelsix, 2018-05-31:
Cool. On a slight side-note there is a new WPF blank template in the templates now if you wanted to make use of that also https://marketplace.visualstudio.com/items?itemName=AngelSix.soliddna-blank