angelsix/solidworks-api #33: CommandManager.RemoveCommandGroup method does not delete command groups

doronkind, 2018-06-01:
SolidWorksEnvironment.Application.CommandManager.RemoveCommandGroup does not remove commands. 

I have two static methods, one to AddCommand groups and 1 to delete:

        private static CommandManagerGroup _partCommandGroup;
        private static CommandManagerGroup _assemblyCommandGroup;

        public static void AddCommands()
        {
            var location = Path.GetDirectoryName(typeof(SolidDnaPlugin).Assembly.Location) ?? string.Empty;
            var iconsPath = Path.Combine(location, @"Assets");

            _partCommandGroup = SolidWorksEnvironment.Application.CommandManager.CreateCommands(
                title: "Parts Export",
                items: new List<CommandManagerItem>
                {
                    new CommandManagerItem
                    {
                        Name = "Upload.Part",
                        Tooltip = "Send to the PrintFromWeb",
                        ImageIndex = 0,
                        Hint = "Send the *.stl model to the Print from Web interface",
                        VisibleForDrawings = false,
                        VisibleForAssemblies = false,
                        VisibleForParts = true,
                        OnClick = Exporting.UploadStl
                    },
                    new CommandManagerItem
                    {
                        Name = "Export.Part",
                        Tooltip = "Export the model",
                        ImageIndex = 1,
                        Hint = "Export the *.stl assembly",
                        VisibleForDrawings = false,
                        VisibleForAssemblies = false,
                        VisibleForParts = true,
                        OnClick = Exporting.ExportStl
                    }
                },
                iconListsPath: Path.Combine(iconsPath, @"commands_{0}.png"),
                hint: "Export parts in formats",
                tooltip: "Such as STL and STEP");

            _assemblyCommandGroup = SolidWorksEnvironment.Application.CommandManager.CreateCommands(
                title: "Assembly Export",
                items: new List<CommandManagerItem>
                {
                    new CommandManagerItem
                    {
                        Name = "Upload.Assembly",
                        Tooltip = "Send to the PrintFromWeb",
                        ImageIndex = 0,
                        Hint = "Send the *.stl assembly to the Print from Web interface",
                        VisibleForDrawings = false,
                        VisibleForAssemblies = true,
                        VisibleForParts = false,
                        OnClick = Exporting.UploadStl
                    },
                    new CommandManagerItem
                    {
                        Name = "Export.Assembly",
                        Tooltip = "Export",
                        ImageIndex = 1,
                        Hint = "Export the *.stl assembly",
                        VisibleForDrawings = false,
                        VisibleForAssemblies = true,
                        VisibleForParts = false,
                        OnClick = Exporting.ExportStl
                    }
                },
                iconListsPath: Path.Combine(iconsPath, @"commands_{0}.png"),
                hint: "Export assemblies in formats",
                tooltip: "Such as STL and STEP");
        }

        public static void RemoveCommands()
        {
            SolidWorksEnvironment.Application.CommandManager.RemoveCommandGroup(_partCommandGroup);
            SolidWorksEnvironment.Application.CommandManager.RemoveCommandGroup(_assemblyCommandGroup);
        }

Adding wokks fine, but remove command does not clear specified groups. Before deleting:

[image]

and after: 

[image]

as you can see, the set of groups is the same.

angelsix, 2018-06-01:
Does the items actually get removed in SolidWorks and its just the references in the `Items` that needs fixing?

doronkind, 2018-06-01:
Unfortunately not. The items still visible in the SW interface

angelsix, 2018-06-01:
I think I remember this being an issue with the API itself. I'll try and investigate shortly.

angelsix, 2018-06-03:
Looks like many years ago I found it as a bug https://forum.solidworks.com/thread/41131 and I think its still there.

In SolidDNA calling the Remove 3 times removes the command group, but leaves other bits. So I dropped to basic code in the `ConnectedToSolidWorks` function and got everything adding then removing once SolidWorks has been open for 10 seconds (so quickly open a Part to see the tab/toolbar then see it go).

The remaining bug is, I can remove the menu, the tab, tab box and command group... but the toolbar created via `HasToolbar = true` cannot be removed. I'll keep playing and if I cannot get that to remove I will report a bug to SolidWorks

```
// Attempt to create the command group
var manager = Application.CommandManager.UnsafeObject;

// Create command group
var errors = 0;
var unsafeCommandGroup = manager.CreateCommandGroup2(5, "Test", "Some tooltip", "Some hint", -1, true, ref errors);

// Add some items
var mainItemID1 = 6;
var mainItemID2 = 7;
var menuToolbarOption = (int)(swCommandItemType_e.swMenuItem | swCommandItemType_e.swToolbarItem);

var cmdIndex0 = unsafeCommandGroup.AddCommandItem2("CreateCube", -1, "Create a cube", "Create cube", 0, "CreateCube", "", mainItemID1, menuToolbarOption);
var cmdIndex1 = unsafeCommandGroup.AddCommandItem2("Show PMP", -1, "Display sample property manager", "Show PMP", 2, "ShowPMP", "EnablePMP", mainItemID2, menuToolbarOption);

// Have toolbar and menu
unsafeCommandGroup.HasToolbar = true;
unsafeCommandGroup.HasMenu = true;

// Activate
unsafeCommandGroup.Activate();

// Remove existing tab
var existingTab = manager.GetCommandTab((int)swDocumentTypes_e.swDocPART, "TestTab");
if (existingTab != null)
{
    manager.RemoveCommandTab(existingTab);
    Marshal.ReleaseComObject(existingTab);
    existingTab = null;
}

// Add tab
var commandTab = manager.AddCommandTab((int)swDocumentTypes_e.swDocPART, "TestTab");
var commandBox = commandTab.AddCommandTabBox();

// Add a command
var commandIds = new int[1];
var commandTextTypes = new int[1];

commandIds[0] = unsafeCommandGroup.get_CommandID(cmdIndex0);
commandTextTypes[0] = (int)swCommandTabButtonTextDisplay_e.swCommandTabButton_TextHorizontal;

var addCommandResult = commandBox.AddCommands(commandIds, commandTextTypes);

// Wait 10 seconds
Task.Delay(10000).ContinueWith(f =>
{
    // Try and remove menu/toolbar has no effect after Activate()
    // and actually stops it being able to be removed...
    //unsafeCommandGroup.HasMenu = false;
    //unsafeCommandGroup.HasToolbar = false;

    // Removing the group works to remove the commands and grey out buttons
    manager.RemoveCommandGroup2(5, false);

    // Now remove the tab
    commandTab.RemoveCommandTabBox(commandBox);
    manager.RemoveCommandTab(commandTab);

    // Unable to remove the Toolbar added via "HasToolbar = true"
});
```

[image]

doronkind, 2018-06-04:
And we can't use iconListsPath in this case, yes?

doronkind, 2018-06-04:
I've found a workaround but anyway it looks not so reliably. It can be considered like a temporary solution:

        public static void RemoveCommandsUnsafe()
        {
            var manager = Application.CommandManager.UnsafeObject;

            // I have just 2 groups, so this hardcode works
            manager.RemoveCommandGroup2(0, false);
            manager.RemoveCommandGroup2(1, false);

            var partTab = manager.GetCommandTab(1, "Name 1");
            manager.RemoveCommandTab(partTab);

            var assemblyTab = manager.GetCommandTab(2, "Name 2");
            manager.RemoveCommandTab(assemblyTab);
        }

angelsix, 2018-06-04:
That just does what I show above, it removes the tab. However you will still have a menu as shown in my screenshot that we cannot remove