angelsix/solidworks-api #34: Export part/assembly as *.stl

doronkind, 2018-06-04:
Does the SolidDNA API provides as opportunity to get file stream in *.stl format. Right now I use Application.ActiveModel.SaveAs method to save *.stl file and read stream after. It works but It would be great to have an oppurtunity to get *.stl in memory. 

    public static async void UploadStl()
        {
            var location = Path.GetDirectoryName(typeof(SolidDnaPlugin).Assembly.Location);
            if (location == null)
            {
                SolidWorksEnvironment.Application.ShowMessageBox("Can't open directory", SolidWorksMessageBoxIcon.Stop);
                return;
            }

            var tempFileName = $@"{location}\temp_{Guid.NewGuid()}.stl";

            var tempExport = SolidWorksEnvironment.Application.ActiveModel.SaveAs(tempFileName,
                options: SaveAsOptions.Silent | SaveAsOptions.Copy | SaveAsOptions.UpdateInactiveViews);

            if (!tempExport.Successful)
            {
                ShowErrorResult();
                return;
            }

            FileUploadResult result;
            using (var stream = File.Open(tempFileName, FileMode.Open, FileAccess.Read, FileShare.ReadWrite))
            {
                result = await MakerbotApi.TryUploadFileAsync(stream);
            }
    }

angelsix, 2018-06-04:
As it just makes use of the underlying SolidWorks API, I don't believe there is an option like that.

closed 2018-06-04 by angelsix

doronkind, 2018-06-04:
Thank you for the explanation. I've investigated SW API, but unfortunately didn't find any suitable options.