angelsix/solidworks-api #35: Export assembly as single *.stl file

doronkind, 2018-06-04:
Does the SolidDNA framework provides an opportunity to save Assembly as a single *.stl file? Right now I can export ASM, but as a result a get several files, according to included parts. 

End can I select the particular part of the assembly and get it in code, e.g. Application.ActiveModel.SelectedPart

angelsix, 2018-06-04:
You just need to set the export options http://help.solidworks.com/2012/english/api/swconst/filesaveasstloptions.htm. For now use `Application.UnsafeObject`. I will bake those options in in future.

For selected part, right now you can do `ActiveModel.SelectedObjects(selectedParts => { /* Code here */})` to get selected objects and check the type to do whatever you want

closed 2018-06-04 by angelsix

ddydeveloper, 2018-06-05:
It is a good suggestion, but I can't understand how can we call SaveAs method for the particular part of the assembly:

[image]

angelsix, 2018-06-06:
You have to do that bit yourself at the minute. Check the type of selected object, which should be swSelCOMPONENTS , and if it is, cast it to an IComponent2, then  do GetModelDoc2(), then save it.

Presuming the only item selected is a part, and not bothering disposing etc... here is an example of getting the model

```
Application.ActiveModel.SelectedObjects(objects =>
{
    var component = objects[0] as IComponent2;
    var model = component.GetModelDoc2();
});
```

doronkind, 2018-06-08:
Unfortunately, It doesn't work as expected. 

1. I try to select particular part: 
[image]

2. Push the Upload button
3. In code handle this part in this way:

        public static List<IModelDoc2> GetAssemblyParts(this Model activeModel)
        {
            var result = new List<IModelDoc2>();

            activeModel.SelectedObjects(objs =>
            {
                foreach (var obj in objs)
                {
                    if (obj.ObjectType != swSelectType_e.swSelCOMPONENTS)
                    {
                        continue;
                    }

                    if (!(obj.UnsafeObject is IComponent2 component))
                    {
                        continue;
                    }

                    result.Add((IModelDoc2)component.GetModelDoc2());
                }
            });

            return result;
        }

4. Call save as method:

[image]

I don't know why, but all of the assembly parts downloaded in separated *.stl files.

angelsix, 2018-06-08:
That will happen with what you are doing. If you want to export the entire assembly as a part just do a saveas on the assembly and save it as a part, then open that file and save it as an stl