angelsix/solidworks-api #37: SetupModelEventHandlers() hooking into events more than once

mtbayley, 2018-06-29:
I tend to get this when changing configurations.

After a long session, there are hundreds of event handlers hooked causing SW to slow down.

For now, this is what I have done. Not sure if this is good practice.
                case ModelType.Assembly:
                    AsAssembly().ActiveConfigChangePostNotify -= ActiveConfigChangePostNotify;
                    AsAssembly().DestroyNotify2 -= FileDestroyNotify;
                    AsAssembly().FileSavePostNotify -= FileSaveNotify;
                    AsAssembly().UserSelectionPostNotify -= UserSelectionPostNotify;
                    AsAssembly().ClearSelectionsNotify -= ClearSelectionsNotify;
                    AsAssembly().ActiveConfigChangePostNotify += ActiveConfigChangePostNotify;
                    AsAssembly().DestroyNotify2 += FileDestroyNotify;
                    AsAssembly().FileSavePostNotify += FileSaveNotify;
                    AsAssembly().UserSelectionPostNotify += UserSelectionPostNotify;
                    AsAssembly().ClearSelectionsNotify += ClearSelectionsNotify;
                    break;
and so on...

angelsix, 2018-06-30:
Oh yeah, slightly over-exited event hooking 😃 

See if this fixes it and everything still works https://github.com/angelsix/solidworks-api/commit/e015f4b9b347db49b1f0b12e67f96cf83d2e2889. You can just update your NuGet package to 1.0.7.2 if you are using NuGet.

mtbayley, 2018-06-30:
I am using VS 2015 at work, and for some reason they don't want to update to 2017. What is the best way to use SolidDna? I like being able to see the source code, but it looks like your project is no longer compatible with VS 2015.

Currently I am using SolidDna code from Nov/Dec last year and haven't updated since.

angelsix, 2018-07-01:
There really is no reason not to use VS 2017. Just run the free version. I simply wouldn't recommend any way to use it in VS 2015. Just run 2015 and 2017 side by side and use 2017 for SolidDna

closed 2018-07-01 by angelsix

mtbayley, 2018-07-01:
Ok thanks