angelsix/solidworks-api #38: Question: Addins dont load into SW and i don't see any errors?

bmanners, 2018-07-03:
I am most likely missing something obvious..but when i run the addin (Custom properties sample or a new project using the templates) the addins don't load in SW. I added the start external program to the project and tried running VS with admin rights but still not happening. Note: I am using SW2016.

brinkdinges, 2018-07-03:
Have you registered the add-in with the SolidDna Add-in installer? That adds the add-in to the Windows register.

Is the add-in available in the SolidWorks list of add-ins? You might have to enable it.

bmanners, 2018-07-03:
I hadn't registered the Add-in with the SolidDna Add-in installer so that helped. I compiled and ran the Addin Installer (under the Tools folder) and registed the addin. I can see the Add in load 
 in the splash screen as Solidworks starts up but it crashes SW now. I will close this (if I am allowed) and start a new issue if i cant fix the crash.

cheers
Brett

closed 2018-07-03 by bmanners