angelsix/solidworks-api #39: How to Add PropertyManagerPage?

emersonbottero, 2018-07-04:
I'm trying to make an abstract class for PropertyManagerPage but I can't get the swAddin as in the Solidworks Template where you get it like that

```
public bool ConnectToSW(object ThisSW, int cookie)
        {
            iSwApp = (ISldWorks)ThisSW;
```

I try to (ISldWorks)Application but it fails to cast on run time.
I was trying to modifying the Exporting tutorial files.

the Ideas are:
1- set standards options like Visible and AligneLeft for control and Visible and Expanded for groups 
2 - create an easy way to add controls to be just something like 
`textbox1 = group1.AddTextBox(textbox1ID,"MyCoolFeature","This is an example textbox")`
instead of
`textbox1 = (IPropertyManagerPageTextbox)group1.AddControl(textbox1ID, controlType, "MyCoolFeature", align, options, "This is an example textbox");`
3 - Get the control cast as an extension like
`Button1.Control.Top = 0;`
instead of
`((IPropertyManagerPageControl)Button1).Top = 0;`

It would be really cool if you could controls the Id's By an Enum and add grammatically  an new Element based on the variable Name, I think it would be like ` group1.AddTextBox(out textbox1,"MyCoolFeature","This is an example textbox") `and it created and textboxID enum.

angelsix, 2018-07-04:
Yeah I will add support for PMP once I get time. I will keep this open until then but it is something I want to do

emersonbottero, 2018-07-04:
Is there a way to use PMP in the conventional way in the meanwhile?

angelsix, 2018-07-05:
Just access the UnsafeObjects of whatever object contains the bits you need. Until I get time to do it, you can do all things in the API as you normally would by just accessing the underlying UnsafeObject interfaces. Just remember if you do that and create any child COM objects to release them once done

emersonbottero, 2018-07-06:
It Worked. 

```
public override void ConnectedToSolidWorks()
        {
            #region Add PMP

            var AppObject = (SldWorks)Application.UnsafeObject;

            var mYPage = new MyPMP(AppObject, "Test");
            var mYPage2 = new MyPMP2(AppObject, "Test 2 even cooler");

            #endregion

            #region AddCommands

```
and the derived class is like

```
        enum Ids
        {
            Group1ID,
            Group2ID,
                Button1ID,
                Button2ID
        }

        PropertyManagerPageGroup mGrupo1, mGrupo2;
        PropertyManagerPageButton mButton1, mButton2;

        #region AddControls

        public override void AddControls()
        {
            //Grupo1
            mGrupo1 = Page.AddGroup(Ids.Group1ID, "test group 1");

            //Grupo2
            mGrupo2 = Page.AddGroup(Ids.Group2ID, "test group 2");
            mButton1 = mGrupo2.AddButton(Ids.Button1ID, "Button 1");
            {
                mButton1.Control().Width = 30;
                mButton1.Control().Top = 160;
                mButton1.Control().Left = 70;
            }

            //Page
            mButton2 = Page.AddButton(Ids.Button2ID, "Button 2");
            {
                mButton2.Control().Width = 30;
                mButton2.Control().Top = 80;
                mButton2.Control().Left = 0;
            }
        }

        #endregion

       Constructor region

       EventHandler region
```

emersonbottero, 2018-07-09:
Here is the Changes I made, probably you can use most of the code.

https://github.com/emersonbottero/solidworks-api/tree/master

closed 2022-10-21 by emersonbottero