angelsix/solidworks-api #40: Add support to FileSaved Event

emersonbottero, 2018-07-09:
I know the framework already has an ActiveFileSaved event but I don`t know if a FileSaved event would be to heavy to process, imagine an open assembly with a lot of external referenced sketch and one change in the assembly changes a hundred parts and of course each one would have been saved even when they are not the active ones.

The real life use I can imagine is that every client has its own rules for their standards like every fillet should be painted in blue, or if the file is in that folder you have to set a custom properties. What I do now is change the save command to my command that check the conditions, update values and then saves. But it only works for active files.

angelsix, 2018-07-10:
Have you checked the `FileSavePostNotify` event we hook into already? https://github.com/angelsix/solidworks-api/blob/develop/SolidDna/AngelSix.SolidDna/SolidWorks/Models/Model.cs#L185 Does that fire for all files? If not, try hooking into `AutoSaveNotify` and see if that fires.

emersonbottero, 2018-07-11:
The Problem is How to hook.
when you hook Application.FileOpened and open and Assembly only the assembly is hooked, it doesn't fire for any component opened together with it. So I can't use the arg2 in "private void Application_FileOpened(string arg1, Model arg2)".

It is like when it is an assembly I had to get every component recursively, check if it has already a hook and then hook it.

angelsix, 2018-07-12:
Yeah I see the issue. Perhaps hook the SolidWorks Application FileOpened event and FileClosed event and keep references to all open files in a list, hooking then into any events you like?

emersonbottero, 2018-07-12:
Yes, but should that be in the framework?

angelsix, 2018-07-12:
Sure. If you get a nice solution I can merge it into the framework