angelsix/solidworks-api #41: SolidWorks Addin Installer

emersonbottero, 2018-07-09:
it says "Unexpected Response" when the computer language is not english. 
But it install the dlls.
Maybe a Localization string are needed...

angelsix, 2018-07-10:
Can you provide me details. Screenshots at least and ideally your compiled project and dll so I can try on my machine

emersonbottero, 2018-07-10:
I try on the Export example you provide in tutorials
[image]

it Should be "Êxito" the Text values translates wrongly.

angelsix, 2018-07-10:
Yeah the localisation right now is English. It should fallback to English I guess instead of null. I will get this fixed

angelsix, 2018-07-11:
Fixed https://github.com/angelsix/solidworks-api/commit/f1e84aae34b1ce4a2e585ddf8a56a8c343a7320c

closed 2018-07-11 by angelsix

emersonbottero, 2018-07-11:
Confirmed in tests. Looks Great Now.