angelsix/solidworks-api #43: Add FirstFeature and NextFeature to ModelFeature class

emersonbottero, 2018-07-10:
I misunderstood or I have to set every time the ModelFeature with new ...

```
namespace AngelSix.SolidDna
{
    public static class Part
    {
        public static void Traverse(Model model, string typeName, Action<ModelFeature> action, bool traverseSubFeatures = false, bool fastTraverse = true)
        {
            
            var CurrentFeature = (Feature)model.AsPart().FirstFeature();
            var Feat = new ModelFeature(CurrentFeature);

            if (fastTraverse)
            {
                
                while (CurrentFeature?.GetTypeName() != "OriginProfileFeature")
                {
                    CurrentFeature = (Feature)CurrentFeature.GetNextFeature();
                }
            }

            while (CurrentFeature != null)
            {
                Feat = new ModelFeature(CurrentFeature);
                //check if it is a specific feature type
                if (Feat.FeatureTypeName == typeName )
                {
                    action(Feat);
                }
                CurrentFeature = (Feature)CurrentFeature.GetNextFeature();
            }
        }
    }
}
```

angelsix, 2018-07-11:
Added to `1.0.1.1` https://github.com/angelsix/solidworks-api/commit/c47313b630e6660ac0a811d7be009a650a2b7953. Update Nuget package then do something like this:

```
Application.ActiveModel.Features((feature, depth) =>
{
    if (feature.GetTypeName2() == "ProfileFeature")
    {
        var sketchFeature = feature.GetSpecificFeature2() as Sketch;

        var segments = sketchFeature.GetSketchSegments() as object[];

        if (segments != null)
            foreach (SketchSegment segment in segments)
            {
                if (segment.GetType() == (int)swSketchSegments_e.swSketchARC)
                {
                    var curve = (Curve)segment.GetCurve();
                                
                    if (curve.IsCircle())
                    {
                        segment.Select4(true, null);
                        var details = (double[])curve.CircleParams;
                    }

                }
            }
    }
});
```

angelsix, 2018-07-11:
Or 

```
Application.ActiveModel.Features((feature, depth) =>
{
    // Wrap any error
    SolidDnaErrors.Wrap(() =>
    {
        // Create feature
        using (var model = new ModelFeature((Feature)feature))
        {
                        
        }
    }, ...);
```

closed 2018-07-11 by angelsix