angelsix/solidworks-api #44: Internal exception thrown "RaceOnRCWCleanup"

doronkind, 2018-07-12:
After updrading from 1.0.0.9 to 1.0.1 version I am getting a following warning: 

> Managed Debugging Assistant 'RaceOnRCWCleanup' : 'An attempt has been made to free an RCW that is in use.  The RCW is in use on the active thread or another thread.  Attempting to free an in-use RCW can cause corruption or data loss.'

There are no any changes in thread usages or something else. Just the version updrading.

angelsix, 2018-07-12:
Gotta love SolidWorks. Fix one thing break another. 

Fixed now https://github.com/angelsix/solidworks-api/commit/32359c22cec9027def525591b7248757164ebf12

Was due to the fact SolidWorks doesn't fire a FileSave event ever, anywhere, if you create a new part. I can also find no events anywhere that fire during that case. If I have a fix here https://github.com/angelsix/solidworks-api/commit/32359c22cec9027def525591b7248757164ebf12#diff-eb2d2b8afc4dda8e379b2faecad78195R673

Nuget 1.0.1.4 has the fix in so just update

closed 2018-07-12 by angelsix

doronkind, 2018-07-12:
Thank you, I'll try to update and notify you if something get wrong.