angelsix/solidworks-api #49: Empty active model

doronkind, 2018-07-24:
Do you provide any API methods to get understand whether ActiveModel has any objects or not. Something like IsEmptyModel method.

angelsix, 2018-07-24:
No but I  guess we could do it fairly easy using the Features() method and based on feature type. What do you mean empty, would you define a part with a sketch as empty? Or a part with nothing at all. Are we talking bodies in parts, and parts in assemblies or something else?

doronkind, 2018-07-24:
I mean we should know is a part just blank, like you just create it and do not add anything. And the same for an assembly.

The high level task is to define can we print the model or not, so if it i s empty we should notify a user.

angelsix, 2018-07-24:
Yeah I think thats a very specific definition. A newly created part still has Axis, History, Planes, Cameras etc.. what defines empty is subjective. You can use

```
ActiveModel.Features((feature, depth) =>
{
    if (feature.FeatureType == ...)
});
```

To determine if any features are what you class as not-empty type features to flag up if its empty or not.

Other than that, I could add definitions such as `ActiveModel.HasBodies()` or something but I think those require specific issues opening for feature requests as this one is a little too vague

closed 2018-07-24 by angelsix

doronkind, 2018-07-24:
Could you specify the version with available Features:

[image]

angelsix, 2018-07-24:
Latest code. Version 1.0.1.4 NuGet package.

doronkind, 2018-07-24:
Thank you, got it! There are 149 feature types, are there any mechanisms to differentiate them, I mean to define that this feature is a body feature and a model can be printed

angelsix, 2018-07-24:
I would just make helper method on the `Model` such as 

```
public bool IsBody()
{
    var isBody = false;
    Features((feature, depth) =>
    {
        if (!isBody)
            isBody = feature.FeatureType == x || feature.FeatureType == y || feature.FeatureType == z;
    });
}
```

And you could make the or check its own function to split the logic.

doronkind, 2018-07-24:
Actually I am going to implement the same, but I'am not sure about the final list of body FeatureTypes, I mean types to define that the model has body (x, y, z etc). And I didn't find SW documentation with explanation. 

I can suppose, that we have default 5 types which are always included (straight the after part/assembly creating). But what if we have more types, which can be added but not related with body?

angelsix, 2018-07-26:
Yeah thats my point, its really a user-defined specification