angelsix/solidworks-api #50: Unexpected behaviour after ActiveModel.SaveAs method calling

doronkind, 2018-07-24:
I've implemented SaveAs method to create a copy of the file for the following operations

`Application.ActiveModel.SaveAs(tempFileLocation, options: SaveAsOptions.Silent | SaveAsOptions.Copy);`

It works as expected with a part: if I select any details the file copy contains just them and it contains a whole model if particular details are not selected.

But in assembly case I got an unexpected behavior: If I try to select particular details (not parts) the file copy is not created. And the result is succesfull, it looks strange. 

[image]

[image]

[image]

[image]

angelsix, 2018-07-24:
I don't get that issue. I select say a Boss inside a part, do SaveAs and it always saves all parts inside the assembly as STLs. Can you share your example file and exactly what it is you are clicking?

doronkind, 2018-07-24:
How can I provide you a short video with repro steps?

angelsix, 2018-07-24:
Just download a free screen capture https://screencast-o-matic.com

doronkind, 2018-07-27:
@angelsix I've created a repository with a test plugin you can reproduce the error:
https://github.com/doronkind/sw-addin-file-access-error

I also have prepared a short video with repro steps:
https://screencast-o-matic.com/watch/cFitQgFLjj

angelsix, 2018-07-27:
Thanks. I will look into this when I get time. Little busy now for good few days but will get around to it