angelsix/solidworks-api #51: Stuck disposing SolidWorksApplication

mtbayley, 2018-07-24:
I get this problem randomly, usually on assemblies when I try to close solidworks with a model open (do not close file first). It seams that ActiveModel_Closing() is called before the Disposing can get set to true so it doesn't ReloadActiveModelInformation.
In my addin, I use the RunOnUIThread a few times, so because the taskpane is already disposed, it just hangs at mInvoker.Invoke((MethodInvoker)delegate { action(); }); This causes SolidWorks to appear closed to the user, but it is still running in the background.
[image]

mtbayley, 2018-07-24:
For now I just increased the await Task.Delay from 200ms to 1000ms as a temporary fix

mtbayley, 2018-07-25:
Users are still complaining their SolidWorks isn't shutting down. 1000ms is not enough. We deal with a lot of large assemblies (1000s of parts). Maybe it has something to do with that.

angelsix, 2018-07-26:
Oh this just goes back to the original issue I know of in SolidWorks that there is no way to tell if all documents are closed. I will leave this open as a bug, but right now I actually have no workaround I have found. 

Can you zip me up a sample assembly so I can just open/close/open/close and investigate it a bit more to find a workaround

closed 2018-09-28 by angelsix