angelsix/solidworks-api #53: Improvment to check if solidworks is installed

emersonbottero, 2018-08-09:
Wouldn't be better to check the registry instead if a file exist?
I installed SW in another drive so every time I have to set it in the interface, instead if we check the registry there is only one possible place.

Computer\HKEY_LOCAL_MACHINE\SOFTWARE\SolidWorks\Licenses\Serial Numbers

I don't know if the uninstaller removes that folder or not.

angelsix, 2018-09-28:
For the installer? It doesn't need to actually file SolidWorks its there more of just a hint if your system has it installed.  I can remove that if thats what you mean?

emersonbottero, 2018-09-29:
I have it installed on T: and I had to browser every time.

angelsix, 2018-10-01:
No longer required https://github.com/angelsix/solidworks-api/commit/c90a43cd00383e7b83a0032a0c3da7f67a018b0d

closed 2018-10-01 by angelsix