angelsix/solidworks-api #54: Unhook events when unloading addin

brinkdinges, 2018-08-13:
Hi guys,

My add-in for creating a new drawing with one click uses events, for example when the active model changes, to show a different panel in the task pane. I have noticed that these events stay hooked after the add-in is unloaded. 

Is this behavior expected? What would be the best way to unhook from these events? I already tried:

- Adding TaskPaneIntegration.RemoveFromTaskpane() in DisconnectedFromSolidWorks, but I checked the code and noticed the form isn't destroyed.
- Adding a destructor to the form and to its view model. These only get called when SolidWorks closes.

Any ideas how to best handle this?

mtbayley, 2018-08-15:
I have noticed this issue as well. Luckily we keep the add-in always enabled, so this isn't really a problem for us.

angelsix, 2018-09-28:
Can you provide some sample code/add-in?

brinkdinges, 2019-05-30:
I finally made an example: https://github.com/brinkdinges/solidworks-api-events-example/commits/master 

It shows an popup every time the active model info changes. When you load the add-in, then unload it, the popup still shows up. It's only not shown when the last document is closed.

brinkdinges, 2019-12-24:
I've been looking into this issue again and I think we shouldn't try to solve it in SolidDna. 

I assume most add-ins will use the ActiveModelInformationChanged event or one of the other events in SolidWorksApplication. We cannot clear the InvocationList for these events, because multiple add-ins might use SolidDna and we would have to remove certain events. We also don't know anything about their WPFControl, so we cannot set the DataContext to null or something like that (if that would even unhook the events).

I have solved it for my add-in by storing the task pane view model instance. When DisconnectedFromSolidWorks is called, I call a ShutDown method on the instance to unsubscribe from the events.