angelsix/solidworks-api #61: Adding more events, useful for others as well?

brinkdinges, 2018-12-16:
I have just added a few more event listeners to my add-in Drew for automatic drawing creation. Would it be useful for other if I add them to SolidDna as well? Or have you kept the amount of events small on purpose? Is the goal to support all active SolidWorks events?

I added:
- ModifyNotify
- RegenPostNotify
- FileSaveAsNotify2
- FileSavePostCancelNotify

If you want to add more, I'm happy to help.

SolidworksDev, 2019-04-25:
I would be interested in having an event listener for OnFileNew event.  Can you point me to the files you had to modify to add these event listeners?

brinkdinges, 2019-04-26:
I didn't implement it in SolidDna yet, just in my own app. I just added:

'SwApp.FileNewNotify2 += NewFileCreated;'

With SwApp being your SolidWorks instance. You could also add it to SolidDna in SolidWorksApplication.cs. 

The events that I added were model events, so you can add them to Model.cs in SolidDna or to a file in your project.

SolidworksDev, 2019-05-16:
Thanks Peter,

I was able to get my event handler working with your help.

closed 2020-05-12 by angelsix