angelsix/solidworks-api #67: COM Object that has been separated from its underlying RCW cannot be used

Venzentzheng, 2019-05-30:
I have 2 add-ins installed in SolidWorks 2016, one is developed without using SolidDNA when I run both add-ins together, I got the error message above, and the exception was thrown when calling ModelDoc2Class.FirstFeature() from the app not developed by SolidDNA, after uninstalling the app developed by SolidDNA, that app works fine again. There are no issues to the SolidDNA-developed app with or without that app. Any ideas on what happened?

angelsix, 2019-05-31:
It could be with us calling FinalReleaseComObject, in the SolidDnaObject. You could reference the source of SolidDna instead of the NuGet, change the ModelExtension class to implement SharedSolidDnaObject instaed of SolidDnaObject, and see if the error goes away. If so we could change them to shared objects so they don't final release.

Venzentzheng, 2019-06-04:
@angelsix Thanks for the reply, I have been testing by following your suggestion, it did not work with the replacement to the ModelExtension, but after I replaced SharedSolidDnaObject on Model class, it's working. shall I replace all classes' SolidDnaObject to SharedSolidDnaObject in your opinion ?

angelsix, 2019-06-04:
No we should only do that for COM objects that the application provides to us, and are not "created" for us. For example `CreateTaskpaneView` even though it comes from the Application is still created specifically for us, and we need to final com release it. Generally the rule of thumb is anything coming from SolidWorksApplication directly that is not a Create method should be a shared object. I've fixed the model one here https://github.com/angelsix/solidworks-api/commit/1754814226241a268f2605003fa0218be64a90a4.

Perhaps the ModelExtension should be shared too, and we could prove it via testing. Can you test in your scenario when you normally get the disposed warning before for Model, with that fixed, try accessing now the Extension and calling something like GetMassProperties() and see if that throws. Then if it does, change it to a shared object and see if it fixes it. If so I'll update the ModelExtension too.

Venzentzheng, 2019-06-04:
sure, I will give it a go, and thanks a lot for the efficiency.

closed 2019-06-04 by Venzentzheng