angelsix/solidworks-api #79: User has to run Solidworks as admin in order to activate the add-in

Venzentzheng, 2019-06-05:
I have created an MSI for my add-in developed upon SolidDNA, after installing it in directory C:\\Directory Files\TargetFolder\MyAddin, I can not activate the add-in with non-admin SolidWorks instance. However, if I install the add-in on D disk, it is ok. any thoughts?

angelsix, 2019-06-05:
If you want a user-specific install (not system wide) you can do this https://github.com/Helmut-Ortmann/EnterpriseArchitect_hoTools/wiki/AddinWithoutAdmin#per-user-installation

To be honest I've not really tested on a setup with an admin and normal user. However if your Wix installer adds the registry entries or runs regasm during install it should work just fine after that without admin rights

Venzentzheng, 2019-06-06:
@angelsix unfortunately, the change did not work. regardless of the admin permission. when I try to run your installer application on the target machine (a virtual machine), it even report errors after I "Install" the application DLL, when I go into SolidWorks, I saw a SolidDNA add-in installed, I think that is your default add-in correct?

Venzentzheng, 2019-06-07:
@angelsix Hi, There is one thing I just discovered quite interestingly, which is also related to the issue in this ticket: when I created a brand new SolidDNA project with blank task pane, I just compiled the project without modifying anything, then I copied everything from the debug folder of the project to C:\Program Files\TestFolder and ran the **SolidWorksAddinInstaller.exe** pointing to the correct DLL of the test project. 

When I ran SolidWorks 2016 not as admin, I can not activate the add-in, but after I ran SolidWorks 2016 as admin, it activates the add-in without a problem. any clues?

angelsix, 2019-06-07:
When you copied the files to Program Files you were admin right? So its stuck with admin rights. I would have to play with it in a setup to make it work right, but it should work fine in theory. If admin installs it and registers it, a normal user has read access so it should work. I'll leave this open and fix it when I get time

Venzentzheng, 2019-06-10:
@angelsix Correct, I am local admin, I can try to fix it and submit a pull request if you could tell me what should I try to fix this issue please ?

angelsix, 2019-06-10:
The issue will be with the Wix installer and file permissions I would say. So its trying to set the permissions of the entire install folder so everyone has read access. However I believe they already do. So thats where I would start to look.

Venzentzheng, 2019-06-10:
```
<Fragment>
         <Directory Id="TARGETDIR" Name="SourceDir">
		<Directory Id="ProgramFiles64Folder">
                        <Directory Id="SubDirManufacturer" Name="$(var.Manufacturer)">
                              <Directory Id="INSTALLDIR" Name="PackagingOptimizer" />
                        </Directory>
	         </Directory>
	</Directory>
</Fragment>

With 
<Package InstallerVersion="200" Compressed="yes" InstallPrivileges="elevated" InstallScope="perMachine" Manufacturer="$(var.Manufacturer)"/>
```

Above is my setup for the installation directory and the package, and I got the reference from [here](https://www.codestack.net/solidworks-api/deployment/installer/wix/) . I am working my hair off to resolve this issue since it is the only one that blocks the add-in going Production.

angelsix, 2019-06-10:
That guide doesn't mention to include

```
    <Property Id="ALLUSERS" Value="1"/>
```

As far as I can see?

Venzentzheng, 2019-06-10:
when I have InstallScope="perMachine" and run the Wix, it tells me the key ALLUSERS has duplicates, but I could not find it even with solution-wide search, when I change it to InstallScope="perUser", it builds fine. but let me try that MSI first with <Property Id="ALLUSERS" Value="1"/> + InstallScope="perUser"

Venzentzheng, 2019-06-10:
ok, did not work well, but I have dragged my entire project into the C:\Program Files\AppFolder now, when I have the project ComVisible set to true, the HEAT.exe stopped me from harvesting something, which did not happen when the project is in D disk. I think I am getting somewhere. finger crossed.

Venzentzheng, 2019-06-10:
Ok, the Wix harvest issue above turns out to be the [ComRegistrationAttribute] was not called or something else, when have a new method declared in  `public class MyAddinIntegration : AddInIntegration` Wix harvest fine again. why it is not calling the ComRegister in SolidDNA? rings any bells?

angelsix, 2019-06-10:
I don't use Heat. I just add the registry entries directly myself. I've never used Heat to do the Com registration

Venzentzheng, 2019-06-10:
@angelsix right, but how do you create an MSI file without using HEAT? and I don't think the problem lies in HEAT anyway.

Venzentzheng, 2019-06-10:
@angelsix because I am installing my add-in on a virtual machine, could you tell me what's the pre-condition in order to set up properly for SolidDNA, please? any frameworks or something like that

Venzentzheng, 2019-06-10:
or is there a way I can find out why SolidWorks is not loading a specific add-in, please? I have googled online, but nothing really helped. thank you

angelsix, 2019-06-10:
Heat just reads a directory and creates components and stuff from it. You can manually just create those entries, or run heat once and then keep that generated file. For the SolidWorks stuff, all regasm does is add registry entries to specific locations. I add those manually in Wix. I need to do a tutorial on it but its not quick or simple to explain here sorry

Venzentzheng, 2019-06-10:
@angelsix thanks a lot for that, good enough to aware that is not the root cause of my problem.. still working on it, I will keep you tuned.

Venzentzheng, 2019-06-11:
@angelsix problem solved, not related to the permission, but conflic between multiple version of Solidworks DLLs, even I set to copy to local and version specific, there is still conflict. so closing the ticket.

closed 2019-06-11 by Venzentzheng