angelsix/solidworks-api #82: Thirdparty data in Model class

Skilllab, 2019-09-27:
Hi Luke!

Is there any options to extend "Model" for reading and writing some class from thirdparty then saving or reading Model? I mean that Model have some "generic" property with  ": where T class"

angelsix, 2019-09-28:
Don't fully understand. Do you mean to save extra data in the actual SolidWorks file so when you send it and re-open it somewhere the data is part of that actual file, or are you talking about some programming specific class extension?

Skilllab, 2019-09-28:
Well, I'm about this:
 ```C#
public class SolidworksModel<T> : SolidworksModel where T : class
    {
        public T ThirdPartyData { get; }

        /// <inheritdoc />
        public SolidworksModel(ModelDoc2 model) : base(model)
        {
            
            ThirdPartyData = new ThirdPartyStorageStoreProcessor().ReadStorage<T>(this, ActiveConfigurationID);
        }

        /// <inheritdoc />
        public SolidworksModel(ModelDoc2 model, string confID) : base(model)
        {
            
            ThirdPartyData = new ThirdPartyStorageStoreProcessor().ReadStorage<T>(this, confID);
        }

        /// <inheritdoc />
        public override int SaveToStorage()
        {
            //Do something

            base.SaveToStorage();
            return 0;

        }
    }
```

But, now I have no clue to get this class from **SolidWorksApplication**. Still working

Skilllab, 2019-09-28:
Maybe something like this:
```C#
public partial class SolidWorksApplication : SharedSolidObject<SldWorks>
    {
//....
        public SolidworksModel<T> GetSolidworksModelExtended<T>() where T: class
        {
            return new SolidworksModel<T>(BaseObject.IActiveDoc2);
        }
//....
    }
```

closed 2019-09-30 by angelsix