angelsix/solidworks-api #88: Upgrade to .NET Core 3

mtrellet, 2020-03-09:
Hello, first thanks a lot for your work for this nuget and whole environment around SW!

Then, our issue:

The official and latest release of the SolidDNA nuget (+ project template) is based on **.NET Framework 4.7.2** which makes the compatibility with existing project based on **.NET Core 3.0+** and **C# 8.0** complicated/impossible...

So, is there any plan to upgrade to **.NET Core 3.0**?

And if this is not in the pipeline, how hard do you think it would be? I guess the SW official API would be one of the limiting factor together with the migration of .NET Framework version of Winforms... There are most likely other limitations but I've put the main ones that come to mind.

Depending on the amount of work you estimate it is, we would be ready to give it a try since it is a possible way to make our software plays nicely with SW. However, we only have limited resources to dedicate to it... (Of course idea would be to share our work here if we succeed!).

Thanks in advance for your help/answer!

angelsix, 2020-03-11:
SoildDNA would be useless as .Net Core. It has to run on Windows and be COM
registered. .Net Core is not COM register friendly. There is also no
benefit to it.

What is your use case where you are using it without that? I guess in a
non-add-in scenario maybe, but again, SW is windows only so I see no use in
.Net Core for it?

mtrellet, 2020-03-11:
Hello @angelsix , thanks a lot for your reply.

I should have stated our issue a bit more clearly indeed. Our aim is to reference a project made in **.NET standard 2.1** to get and use its models, services, etc. directly in the SolidWorks add-in we are developing. Idea was to have a tight pipeline between our software and SW that would manipulate the same kind of objects, in the same way. Linking the two by referencing the Models in the add-in would have allowed us to avoid duplicating and then keep in sync two versions of the same Models (one based on .NET standard 2.1 and the other in .NET framework 4.7.2). 
And just for the record we are indeed using Windows so nothing fancy there.

But before going further you've pointed out something we were not completely aware of:

> .Net Core is not COM register friendly

So this would indeed make things complicated. Anyway, we took the problem the other way and decided that a simpler solution would be to have a file-based communication between the add-in and our project at the end. Less integration but better decoupling between the projects, not too bad at the end. Sorry for the bothering and thanks again for your time and work.

closed 2020-05-12 by angelsix