angelsix/solidworks-api #103: Cannot get localized strings because ConfigureServices is overridden

brinkdinges, 2020-07-04:
I have just updated SolidDna in a project from 1.0.1.9 to 1.0.1.13, the latest version. Now I get an exception early when starting up the add-in. I narrowed it down to the fact that no LocalizationManager is ever set up, so getting a localized string in a SolidDna wrapper fails.

ConfigureServices in my add-in is being called and it's been empty for three years. I now see that on May 13th, this method went from abstract to virtual. Setting the LocalizationManager happens in the virtual base method ConfigureServices within SolidDna. 

Should we keep setting up the LocalizationManager in this method? What if another user overrides this method and then has no LocalizationManager?

dhozen, 2021-11-05:
I've been trying to update to the latest version of SolidDna (1.01.13) because the earlier versions (1.01.10 and earlier) don't accept the 2020 interop files and regasm fails. However, 1.01.13 registers OK with the 2020 interops, but crashes on opening swx. This happens, so far as I can tell, in the ConnectedToSolidworks method.

The method successfully creates the taskpane object and gains access to the mswApp, but crashes before any windows show in SolidWorks.

Could this be related to the issue you describe above?

[image]

brinkdinges, 2021-11-05:
You should use the same DLL version as SolidDna uses so your project compiles and can be registered. It has no influence on how your app runs. See my explanation [here](https://github.com/angelsix/solidworks-api/issues/118).

If you have a ConfigureServices method in your add-in, delete it so it is not overridden with an empty method.

Have you tried a try-catch around your code in ConnectedToSolidWorks? That should give us a hint and hopefully, an exception message.

dhozen, 2021-11-05:
Thanks Brink! Commenting out the whole "ConfigureServices" method fixed it.

And thanks for the DLL version explanation...

-Jason S.

dhozen, 2021-11-05:
Rather thanks Peter :^) Komt u uit Nederland?

brinkdinges, 2021-11-07:
Awesome, you're welcome. And yes, that's where I'm from ;)

dhozen, 2021-11-07:
Ik heb in Zaandam een jaar gewoond (1968). Spreek een beetje, maar heb veel vergeten. Danke nog een keer...