angelsix/solidworks-api #111: Not able to register the Addin

Mussawer, 2020-12-07:
I am not able to register the dll because of Solidworks.interop.swpublished in the references of the project but as I remove the reference from the project I am able to register the dll but the addin is not loaded in solidworks neither able to debug it through visual studio

brinkdinges, 2020-12-16:
Do you have the DLL property "Embed interop types" set to Yes/True by accident? It should be set to No.

Mussawer, 2020-12-19:
I have set it to "No" 
But I have begin my work using another approach
Thanks for your help!!!!

ConyKitty, 2021-09-22:
Hello, Have you fixed the issue of swpublished dll ? I also the all the solidworks.interop.xxxx dll embed interop type to false, it doesn't work either.

brinkdinges, 2021-09-22:
There is no known problem with the DLLs and "it doesn't work" isn't enough for us to help you. Please share a small project that lets someone reproduce the issue and share the exact errors you're seeing.

I can think of three things that can go wrong:
1. Embedding the DLLs is set to true/yes. You have checked for that.
2. You have added DLLs to your own project that are not from SW2021. SolidDna references the SW2021 DLLs and you should too. See https://github.com/angelsix/solidworks-api/issues/118
3. You have no GUID attribute added to your AddInIntegration class. I cannot find the issue where we talked about this.

elincooper, 2021-09-28:
Hello, I have also the same problem with the solidworks.interop.swpublished.dll. I had installed SW2018 (now deinstalled) and now I want  update to SW2021 (installed). So i have test the NuGet. I get this information from the registration:

RegAsm: warning RA0000: If you register an unsigned assembly with the / codebase option, the assembly may interfere with other applications installed on the computer. The / codebase option should only be used with signed assemblies. Give the assembly a strong name and reregister it.
RegAsm: error RA0000: The file or assembly "SolidWorks.Interop.swpublished, Version = 28.1.0.74, Culture = neutral, PublicKeyToken = 89a97bdc5284e6d8" or a dependency on it was not found. The assembly manifest definition found does not match the assembly reference. (Exception from HRESULT: 0x80131040)

The used Version of SolidWorks.Interop.swpublished is 29.4.1.1! I also deleted it in the reference, this has no effect either.

Link to my test project according to the AngelSix video [https://www.youtube.com/watch?v=r8e3xFlmZ_o](url)
[https://github.com/elincooper/SW2021AngelSixNuGetTest](url)

Thanks for your help.

brinkdinges, 2021-09-28:
You are correct. The SW2021 DLLs are not yet in this NuGet package. 

They are in my fork, where I also made some bug fixes that are lingering here as PRs. I also made some big changes, so I'll add some documentation first.
https://github.com/CAD-Booster/solidworks-api

Thanks for the test project. I found out two things:
1. You can get it to register by adding a reference to the SW2020 Published dll.
2. You can get it running by also adding all other SW dlls.