angelsix/solidworks-api #116: Taskpane Freezing

cabuk79, 2021-01-13:
Not sure where I have gone wrong with my code.

Everything has been working fine when debugging but all of a sudden the task pane on the my addin is freezing on the last selected pane.

So if I select another pane such as File Explorer then go back to my addin it shows the File Explorer task pane view.

It looks like it is trying to load but freezes on the last selected tab data.

[image]

I have taken the code right back to starting a new blank SolidDna project and it still gives the same result.

Has anyone else seen this or can guide me in the possible causes.

Thank You

brinkdinges, 2021-02-04:
Related SW forum topic: https://forum.solidworks.com/message/1097545

Can you share your code? It's hard to debug or reproduce anything without any code.

cabuk79, 2021-02-04:
> Related SW forum topic: https://forum.solidworks.com/message/1097545
> 
> Can you share your code? It's hard to debug or reproduce anything without any code.

Hi I decided to remove any reg key I could find to SolidDna not just my addin and this seems to have worked as the task pane is now loading with no issues.

I have no idea what I did here but will as advice in the SW forum start to use git and commit often.

Thank you for your reply I will close this issue off.

closed 2021-02-04 by cabuk79